xoseperez/sonoffsc #7: {Suggestions} Adding support for more environmental sensors

felixekman, 2017-02-15:
A cool addition to this project would be to add support for these sensors which would add further to the SC's usefulness as a home sensor:

AS3935 - A really cool sensor that can sense and warn for a lightning storm within a radius of 40km and trace down to 1km in 14 steps.

MPL3115A2 - Altitude/Pressure Sensor for barometric data and also as an added bonus 12-bit temperature measurements.

Alternatives to the unstable and low life-expectancy of the DHT22

HIH6130 - Humidity and Temperature Sensor
Si7021 - Humidity and Temperature Sensor

xoseperez, 2017-02-15:
I do have an AS3935 so I can try to add support for it. I didn't know about the MPL3115A2, but I was planning to add support for the BME280 (humidity, temperature and pressure). Also wanted to add support for MQ7 carbon monoxide sensors and probably PIRs or microwave sensors.
Wow! That's a lot of sensors... should I create and issue for each one of them?

felixekman, 2017-02-15:
That would be great! AS3935 would be amazing to have connected to my MQTT smart home. Microwave sensors would probably be the easiest to integrate into the device so no modification to the case would have to be made. As for MPL3115A2 vs BME280, I've tested both and they are pretty equal but I've found MPL3115A2 a little more stable.

farconada, 2017-02-17:
a CO2 will be great and useful for your day to day

pdominique, 2017-02-17:
BME280, microwave sensor and PIR? Sounds good, exactly what I wanted to add to the Sonoff SC!

felixekman, 2017-03-11:
I've looked into microwave sensors and there seems to be an issue finding modules that work well with the SC, as most modules demand amplifiers and a lot of extra supporting electronics. As for sensors, I haven't had much success with the HIH6130 and Si7021 anyone else had some progress?