xoseperez/sonoffsc #9: Add option of fan control on pin 7

Justblair, 2017-03-03:
Pin 7 is allocated on the pcb as for an LED.  Another option is to use that circuit for selective fan control to stir the air before making a dust sensor reading.   ie we could digtialWrite(FAN_PIN, HIGH); a few seconds before taking a reading.

xoseperez, 2017-05-13:
What kind of fan would fit into the SonoffSC enclosure?
Maybe this one? https://www.aliexpress.com/item/new-SEPA-MF15B-05-5V-0-06A-1-5cm-15mm-1505-15x15x5mm-mini-micro-fan-server/32617583921.html

Justblair, 2017-05-15:
I managed to squidge one of these in:

http://www.ebay.co.uk/itm/5V-Cooler-Brushless-Micro-DC-Fan-25-10mm-Mini-Cooling-Computer-Laptop-Fan-Black-/252053118110?hash=item3aaf895c9e

Not sure it made much difference, but at the time I was getting no readings at all.  I since turned up the sensitivity of the smoke sensor and managed to get something that looked like a reading.

 

It was a tight squeeze, I seem to remember repositioning the dht22 a little lower to make it all fit.

 

I have no idea where the sensor is at the moment.  I moved house a couple of weeks ago and still have loads left to unpack