xoseperez/sonoffsc #10: Cannot get the ESP code to compile

jaysuk, 2017-03-09:
Using the arduino IDE on windows 10.
Have all the dependencies included in the arduino library.
This is the error I get

C:\Users\live\Downloads\sonoffsc-master\esp8266\sonoffsc\src\sonoffsc\button.ino: In function 'void buttonLoop()':

button:25: error: 'class DebounceEvent' has no member named 'getEvent'
         uint8_t event = _button.getEvent();
                                 ^
button:27: error: 'EVENT_DOUBLE_CLICK' was not declared in this scope
         if (event == EVENT_DOUBLE_CLICK) createAP();
                      ^
button:28: error: 'EVENT_LONG_CLICK' was not declared in this scope
         if (event == EVENT_LONG_CLICK) ESP.reset();
                      ^
exit status 1
'class DebounceEvent' has no member named 'getEvent'

Any ideas?

juancreyes, 2017-03-11:
Hi,

  I am facing same error trying to verify esp8086 set files. I am afraid of button.ino code turned into bug because the Debounce library removed and changed some classes at Jan 25th and likely it misses the proper changes along of button. 

[2.0.0] 2017-01-25
 Changed:
- Reporting # of clicks and length instead of predefined events
- Callback signature changed to report click count and length
Removed:
- EVENT_SINGLE_CLICK, EVENT_DOUBLE_CLICK, EVENT_LONG_CLICK

    I would wait for Xoje to take a look and propose the fix. @xoseperez do you agree with this bug ?.

Regards
Juan

jaysuk, 2017-03-12:
I managed to compile it with the older debounce library.
It now creates an access point that I can connect to but once I log in with the correct credentials, i get an error 404 and cannot go any further to configure the sonoff sc

juancreyes, 2017-03-14:
I dont know what can cause 404 error. Each time you press external Button, do you call AP creation, createAP() ? Does the SSID sonoff_endMAC appear on your wifi list and you can authenticate to AP successfully ?

jaysuk, 2017-03-14:
The sonoff+Mac address appears. I can navigate to espurna.local and it asks
me to login. I use the standard admin and fibonacci login and that's as far
as i can get. The sonoff fails to serve any more web pages so i cannot
configure it with the correct WiFi info etc

xoseperez, 2017-04-02:
Sorry I did not get any notifications on this issue on my email...
@jaysuk you are right, compiling against DebounceEvent 1.1 is the way to go, the code is still not compatible with new 2.0 version.
@jaysuk Probably you did not flash the filesystem image, so there are no contents to serve.  You can check how to do it in the ESPurna repo (same procedure) here: https://bitbucket.org/xoseperez/espurna/wiki/Filesystem.deprecated.md

gbpucci, 2017-04-15:
How i can compile with debounceEvent 1.1 ??

DasBasti, 2017-04-24:
https://github.com/xoseperez/sonoffsc/pull/12 will fix the issue with DebounceEvent