xoseperez/sonoffsc #11: Embedis EEPROM dictionary doen't work with SerialLink

juancreyes, 2017-04-01:
I have connected serial port to ESP8266 side and remove serial communication toward Atmega. I think Embedis seens doesn't work properly with SerialLink.h runing at same time. Tested "commands", "dictionary" and no lucky. However if I put serial terminal "AT+TEMP=230" I got success MQTT response ([MQTT] Sending /device/temperature 226.50).

If my diagnostic is right, Any ideas how to figure out a solution ?.

BR

xoseperez, 2017-04-02:
Embedis commands are disabled because I did find they were not compatible with serial communication with the ATMega328. The parser is handled in the settingsLoop method which is not being called in the main loop.

juancreyes, 2017-04-02:
Thanks for your feedback. It is physical limitation of having just one serial interface.

closed 2017-06-27 by xoseperez