xoseperez/sonoffsc #17: Webinterface not showing up

MikelillOi, 2017-07-12:
Hi!

I flash the ATMEGA328 succesfully and the esp8266 too. But when I try to go to the webinterface it shows the password dialog. I put it, but then, I always received a 404 error. 

I flashed the espurna normal firmware and it works, but I only can process as a sonoff switch.

The console, only shows "[WEBSERVER] Request: GET /" when I received the 404 error.

DerKorb, 2017-09-10:
There is another target "upload spiffs" in platform.io. You need to run it for the webserver content to be uploaded to the esp.

timgray, 2017-09-13:
I also am getting this and I am not using Platform.IO.   No web interface at all.   I get the login request and after that it fails.

DerKorb, 2017-09-13:
you need to upload the webserver content even if you are not using platform.io
no idea how to do it without platform.io

xoseperez, 2017-10-05:
Please note version 1.0 does not use SPIFFS. Instead the web content is converted into a PROGMEM byte array and incloded in the main firmware image. You should have a index.html.gz.h under esp8266/sonoffsc/static. If you don't you can build it again following this [instructions from the ESPurna repository](https://bitbucket.org/xoseperez/espurna/wiki/WebInterface.md).

bastshoes, 2018-01-26:
I'm also can't get to web interface. After inputting default admin password I'm getting wrong credentials message. Browser console log gives 404 error. I'm using latest master branch.

Cabalist, 2018-09-04:
Ditto on the "Wrong Credentials"  this happens after I put in 'admin/fibonacci' into the  WebUI's login.  If I put in 'admin/foo' it correctly rejects that so it seems the issue is after the first round of authentication.  

I imagine at this point the codebase has fallen a good deal behind core espurna.

minipilot1300, 2018-09-23:
Hi,

same for me. Wrong Credentials when i put admin/fibonacci. Any other password reject, the input mask is shown again.

minipilot1300, 2018-09-24:
*** SOLUTION ***
Delete the Memory of the esp. Than flash the firmware. Than the sc will do the job!

I use esptool.py -p /dev/cu.usbserial-00000000 erase_flash for this. It is possible that platform io can arrange this too.

wiatrM, 2020-04-09:
@xoseperez  @minipilot1300 Any other solutions for this problem? Im trying to log-in (also disabled login check in defines) but with lo luck (html Wrong Crenentials displayed even when Im putting right credentials :/)