xoseperez/sonoffsc #19: Can't flash ATMega328P

revdaz, 2017-09-29:
Hello,

I really like your work, but I can’t flash the ATMega328P with Arduino and do not know what I’m doing wrong. I flashed a lot of sonoff devices and also the ESP part of the Sonoff SC, but don’t know how to flash the ATMega328P.
I have compiled everything without errors, but when I want to upload the code I get everytime this message:
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x1c
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x1c
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x8e
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x1c
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x1c
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xfc
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x70
Beim Hochladen des Sketches ist ein Fehler aufgetreten

Have I press a button to get in flash mode? I have choosen Board: Arduino Nano and processor: ATMega328P. The green LEd is blinking on the Sonoff. In the serial monitor I see AT+STATUS?
I have also disconnected the jumpers.
I really tried everything and getting crazy….

Thank you very much for help

xoseperez, 2017-10-05:
I use “Arduino Uno” as board type. Also, you can try flashing it via the serial header using an USB2UART programmer. Check this[ post about the Sonoff SC](http://tinkerman.cat/itead-studio-sonoff-sc-revisited/).

kollaesch, 2017-10-07:
Hi I have the same error here. 

@xoseperez **What is the correct commandline-command to do this?**
Platformio and arduino-IDE should list it. Please post here for reference.
To get it please run `platformio run --target upload -v` in the directory
`atmega328`. Thanks!!

This saves us setting up both systems (with more potential traps)
I'd like to rule out that I killed the ATMega328P.

jrbenito, 2017-10-19:
Hi Guys,

I received my sonoffsc and flashed ESP in few minutes but struggled 2 days with ATMEGA. Nothing worked! I tested ATMega and saw it was not dead by connecting serial terminal and seeing AT+STATUS? from original firmware being sent over (be aware of the baud rate). I first bet reset pin was not working but multi-meter show everything ok.

Finally I downloaded the original source code to try put some light and read something like "first flash bootloader" in the readme file. Immediately thought "Are they flashing with ISP without arduino bootloader?"; grabbed my ArduinoDUE that was with ArduinoISP firmware, wired everyting (3v3, so do not put sonoffsc on the USB or DUE  will fry), burned bootloader and _voilá_, not it flashes ok but fail reading (fuse issue). Worked!

Maybe you have a sample like mine? Mine is a V2 PCB with DHT11 soldered (I removed it and put headers so I can change sensor every time it fails - DHTxx fails a lot).

BR,
Benito

ghiglie, 2017-10-21:
@jrbenito Thanks, I saw your post on comments thread. I'll use my Uno as ISP... I've planned to do it later next week, but now... I want to try!  👍

kollaesch, 2017-10-27:
If I access the 'atmega' by`picocom -b 19200 /dev/tty.usbserial-A50285BI`
I get the following response.
```
AT+STATUS?t is error
it is error
...
```
Ideas? Couldn't find anything in the internet. I don't have any DHT** connected.
So I'm wondering if this is because of the missing DHT or because something 
on the atmega is faulty. (hardware or firmware)

Wieg, 2017-11-05:
Can somebody get me on the way ? i am flashing my SC for the second time de Atmega328p but when i try now i get this error. Who has the solution ?

38588       106    1020   39714    9b22 .pioenvs/sonoffsc/firmware.elf
Error: The program size (38694 bytes) is greater than maximum allowed (32256 by
tes)
*** [.pioenvs/sonoffsc/firmware.elf] Explicit exit, status 1
========================== [ERROR] Took 14.03 seconds =========================
=

jrbenito, 2017-11-05:
Hi,

Your firmware is too big

Wieg, 2017-11-05:
its the standard download 1.1.1.1 from here .... what to do ?

jrbenito, 2017-11-05:
Didn't you add nothing? The firmware size you reported is 38k while
atmega328 only has 32k (31k if you use arduino bootloader).

Wieg, 2017-11-05:
Yes i know ..... 32k ... strange thing is ... i didn't change anything

I tried, download, unzip open atom/platformio connect the atmega328 and flash - Size to big
(flashed the 8266 also with the firmware, no problems at all)

Followed tinkerman and xoseperez .... and get this result ..... Hardware wise i added the DHT22, LED ring, detection sensor .... i am sooooo close in making it all work 

i looked at the libaries ... they are all needed, so i am a bit stuck in the final fase of making it work.

i need some tips from guys who did this project .................

jrbenito, 2017-11-05:
I am far from my computer today to test anything. If you can't solve it
later I will try to reproduce.

In my case I discovered I could not flash atmega328 because it was shipped
without arduino bootloader. After using isp instead of usb/ttl everything
was OK.

VooDooN1, 2017-11-05:
I have Sonoff SC Rev.2.0. I tried to flash ATMega328P for about two months. Tried many options. First I used FTDI USB to TTL with a lot of different drivers. Then I bought another FTDI USB to TTL and again tried different options. Then I bought USBASP and again tried a lot of drivers. Using Platformio and Arduino IDE. Nothing helped. But I was able to reflash ATMega328P only after I updated the firmware on the USBASP, using the following instruction (in Russian): http://xdomus.ru/usbasp-upgrade. After that, everything worked. Hooray!!!

maur8ino, 2017-11-05:
I have exactly the same problem as @Wieg: SonofSC V2.0 w/ DTH11 soldered and no other sensor mounted, freshly downloaded v1.1.1 and getting the error:

```
Error: The program size (38940 bytes) is greater than maximum allowed (32256 bytes)
```

I'll try to downgrade

Wieg, 2017-11-05:
Let me know how it goes @maur8ino .... pulled my hairs out, left it for a moment .... now doing some homeassistant scripts too "clear" my mind

maur8ino, 2017-11-05:
I was able to compile it and flash it only by commenting out all the references to the **Adafruit NeoPixel** and **WS2812FX** which brings down the size to **10176b**.

Since I have no led connected, I don't really need them, but other might need them.

@Wieg if you want I can fork the project and just commit the change I've in local 😄

Wieg, 2017-11-05:
yes please @maur8ino  ..... ill start flashing ..... and thinking about my led solution ....

maur8ino, 2017-11-05:
@Wieg here's the temporary fix https://github.com/maur8ino/sonoffsc!

Wieg, 2017-11-05:
great ... have a beer on me !

GregSutcliffe, 2017-11-05:
@maur8ino @Wieg your timing is awesome, I just went to reflash 1.1.1 to my SC after months of doing nothing with it - hit the "too big" issue, and your patch saves me. Thanks!

Wieg, 2017-11-06:
hahaha i am still thinking ... i built in the presence sensor, dht22 and the LED .... stiil don't get how tinkerman and XO did the firmware for that.

jrbenito, 2017-11-06:
Maybe some library upgraded recently?

maur8ino, 2017-11-06:
@jrbenito mmm I just did a fresh clone and I'm still getting the error

```
Device: atmega328p

Program:   38694 bytes (118.1% Full)
(.text + .data + .bootloader)

Data:       1126 bytes (55.0% Full)
(.data + .bss + .noinit)
```

the size is 118.1% 😢

jrbenito, 2017-11-06:
@maur8ino ,

It is what I just said, I cloned fresh new code and downloaded libraries (platformio does it for you) and got oversized firmware. However, I do believe that code at my home computer compiles so the only thing I can think about is some library update that made code size exploded.

Checking here with fresh cloned code, I checked out later version of @xoseperez  code and give me exact same code size as you got. So, the issue is not in sonoffsc code but in some library. I opened platformio.ini and checked library dependency and:

- Arduino library for DHT11DHT22
last updated two years ago - seems not guilty 

-Adafruit_NeoPixel
last update 15 days ago - maybe guilty

-WS2812FX
it was not released  but had updated 5 days ago, however, if you download fresh new library it will dowload latest commit from git, hence, even not released as new version for platformio, the library code will be download from latest code - maybe guilty (my bet goes to this one)

-Ticker
last update 6 months ago, so not guilty 

So, I clone WS2812FX but checkouted it to a later version (SHA 069139b9340e14cd46d8fafe14972b6c056273ec) and compiled with code size 29948 bytes. Checked out at sha b31c3fbcffa510afe46897cb2261bbcd6c892249 and code size compiled 38694 bytes. Every sha between those above gave me compilation error.

Hence, the guilty is WS2812FX library, if you want to correct it by hand (workaound) do:

cd .piolibdeps/WS2812FX_ID1647
rm *  
mv .libray.json ../
git clone https://github.com/kitesurfer1404/WS2812FX.git .
git checkout  069139b9340e14cd46d8fafe14972b6c056273ec
mv ../.library.json .

Now your code will, hopefully, compile with  manageable size.

BRs,
JrBenito.

jrbenito, 2017-11-06:
Filed an issue to the library owner: https://github.com/kitesurfer1404/WS2812FX/issues/50

Wieg, 2017-11-06:
@jrbenito awsome ........ ill be checking in to this in a few hours ........ can't wait :) great community here .... thank you very much

maur8ino, 2017-11-06:
@jrbenito thanks for the explanation! 😄

maur8ino, 2017-11-06:
@jrbenito BTW you can pin a library to a specific **git** hash like this 😄 

```diff
lib_deps =
    DHT sensor library
    Adafruit NeoPixel
-    WS2812FX
+    https://github.com/kitesurfer1404/WS2812FX#069139b
    Ticker
    https://github.com/xoseperez/seriallink#0.1.0
```

I updated my fork pinning the **WS2812FX** library to the latest working hash.

jrbenito, 2017-11-06:
Perfect

https://github.com/kitesurfer1404/WS2812FX#069139b
    https://github.com/xoseperez/seriallink#0.1.0

Wieg, 2017-11-07:
it worked great .... now to play with it in homeassistant ;)

lorenzodroid, 2018-03-14:
Hi, I've the same problem :/

pablobolomey, 2018-08-12:
@maur8ino I'm trying to use your releases with pio and im getting the foloowing issues when project is open;
\Documents\sonoffsc-1.1.1\esp8266.piolibdeps_tmp_installing-tyamj9-package'... warning: Could not find remote branch 1.1.4 to clone. fatal: Remote branch 1.1.4 not found in upstream origin Error: Traceback (most recent call last):

I this project still alive?

Thanks

Pablo

maur8ino, 2018-08-13:
@pablobolomey I just tried to do a `pio run` on a fresh checkout and I traced the error to `justwifi` which is an external (the only result when searching for **1.1.4** in the project btw): **xoseperez** have removed all the tags and pushed just the `master` branch.

You __could__ get away with removing the `#1.1.4` from `esp8266/platform.ini` like:
```diff
-  https://bitbucket.org/xoseperez/justwifi.git#1.1.4
+  https://bitbucket.org/xoseperez/justwifi.git
```
   
but I dunno if it's gonna work.
I don't have the hardware anymore so I can't try 😢.

Let me know if it worked!

pablobolomey, 2018-08-13:
Thanks, I finally work that out and solved that issue but now it’s shows other complied warnings and errors.

Regards,
Pablo Bolomey

Sent from iPhone

maur8ino, 2018-08-13:
@pablobolomey it the compilation failing?If it's just warning the compilation will most likely to go through.

pablobolomey, 2018-08-13:
Yes it’s failing due to some syntax and veritable errors missing or not define words.

I have downloaded the zip file of the code

Pablo

Sent from iPhone

xoseperez, 2018-08-14:
This issue is fixed in the `dev` branch. If you want to use `master` you should change the library dependencies, but not in the way @maur8ino suggested (he's updating to a newer version of the justwifi library that is not supported in the master code). Instead use this:

```
https://github.com/xoseperez/justwifi.git#1.1.4
```

maur8ino, 2018-08-14:
@xoseperez thanks for the clarification, I didn't realize that you're moving all your repos to Github from BitBucket hehe

pablobolomey, 2018-08-15:
thanks @xoseperez @maur8ino, I have moved to the dev files. corrected some errors but still getting these:

Once again thanks for all the support

Pablo

Compiling .pioenvs\sonoffsc\lib665\ESPAsyncTCP\ESPAsyncTCP.cpp.o

C:/Users/pablo.bolomey/Desktop/sonoffsc-dev/esp8266/sonoffsc/lights.ino: In function 'void sendNotification(bool, long unsigned int)':
C:/Users/pablo.bolomey/Desktop/sonoffsc-dev/esp8266/sonoffsc/lights.ino:175:33: error: 'class Ticker' has no member named 'once'
if (time > 0) _lights_defer.once(time, sendNotification, !state);
^
C:/Users/pablo.bolomey/Desktop/sonoffsc-dev/esp8266/sonoffsc/mqtt.ino: In function 'void mqttSend(const char*, const char*, bool)':
C:/Users/pablo.bolomey/Desktop/sonoffsc-dev/esp8266/sonoffsc/mqtt.ino:154:28: error: 'class Ticker' has no member named 'once_ms'
_mqtt_flush_ticker.once_ms(MQTT_USE_JSON_DELAY, _mqttFlush);
^
C:/Users/pablo.bolomey/Desktop/sonoffsc-dev/esp8266/sonoffsc/ntp.ino: In lambda function:
C:/Users/pablo.bolomey/Desktop/sonoffsc-dev/esp8266/sonoffsc/ntp.ino:84:24: error: 'class Ticker' has no member named 'once_ms'
_ntp_delay.once_ms(100, _ntpUpdate);
^
C:/Users/pablo.bolomey/Desktop/sonoffsc-dev/esp8266/sonoffsc/utils.ino:266:18: error: 'class Ticker' has no member named 'once_ms'
_defer_reset.once_ms(delay, reset, reason);
^
C:/Users/pablo.bolomey/Desktop/sonoffsc-dev/esp8266/sonoffsc/ws.ino: In function 'void _wsParse(AsyncWebSocketClient*, uint8_t*, size_t)':
C:/Users/pablo.bolomey/Desktop/sonoffsc-dev/esp8266/sonoffsc/ws.ino:97:58: error: 'class Ticker' has no member named 'once_ms'
if (strcmp(action, "reconnect") == 0) _web_defer.once_ms(100, wifiDisconnect);
^
Compiling .pioenvs\sonoffsc\lib665\ESPAsyncTCP\ESPAsyncTCPbuffer.cpp.o
Compiling .pioenvs\sonoffsc\lib665\ESPAsyncTCP\SyncClient.cpp.o
Compiling .pioenvs\sonoffsc\lib665\ESPAsyncTCP\tcp_axtls.c.o
Compiling .pioenvs\sonoffsc\libab9\Time\DateStrings.cpp.o
Compiling .pioenvs\sonoffsc\libab9\Time\Time.cpp.o

Archiving .pioenvs\sonoffsc\lib0bc\libESP8266WiFi.a
Compiling .pioenvs\sonoffsc\lib22d\Hash\Hash.cpp.o
Compiling .pioenvs\sonoffsc\lib22d\Hash\sha1\sha1.c.o
*** [.pioenvs\sonoffsc\src\sonoffsc.ino.cpp.o] Error 1
 [ERROR] Took 5.71 seconds
 
 [SUMMARY]
Environment ota         [SKIP]
Environment sonoffsc    [ERROR]
 [ERROR] Took 5.71 seconds

DrCherry, 2019-02-03:
Same, removed support for the LED ring and the size was correct.

CreaturePH, 2019-07-07:
Guys, no one has compiled binary files for firmware (preferably for ATMega too)? Honestly, I'm already a little tired of fighting this device. I bought it about two years ago, and since then it has been lying around disassembled so far due to the fact that I just can not flash it. Every time I try, then I stumble upon any difficulties with compiling and throw it again. Once again, I took it, taking into account all the above-mentioned features, I adjusted the list of dependencies, but again nothing good comes out:
```
In file included from .piolibdeps\NtpClientLib_ID727\src/NtpClientLib.h:50:0,
from sonoffsc\config/prototypes.h:2,
from sonoffsc\config/all.h:4,
from C:/Users/Creature/AppData/Local/Programs/Microsoft VS Code/sonoffsc-master/esp8266/sonoffsc/sonoffsc.ino:23:
C:\Users\Creature\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include/lwip/dns.h:104:54: error: 'ip_addr_t' has not been declared
typedef void (*dns_found_callback)(const char *name, ip_addr_t *ipaddr, void *callback_arg);
^
C:\Users\Creature\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include/lwip/dns.h:108:43: error: 'ip_addr_t' has not been declared
void           dns_setserver(u8_t numdns, ip_addr_t *dnsserver);
^
C:\Users\Creature\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include/lwip/dns.h:109:1: error: 'ip_addr_t' does not name a type
ip_addr_t      dns_getserver(u8_t numdns);
^
C:\Users\Creature\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include/lwip/dns.h:110:56: error: 'ip_addr_t' has not been declared
err_t          dns_gethostbyname(const char *hostname, ip_addr_t *addr,
^
```
Thanks in advance for any of your help.