xoseperez/sonoffsc #20: SerialLink - Allow Char instead Int

levipereira, 2017-10-03:
Hi,
First of all, many thanks for your sharing...it's wonderful.
I'm using your Lib Seriallink.h to others projects with esp8266 and arduino uno.  I need send/receive char over serial comunication. 
Far I can understand, it's allow only int values.
I'm not familiar with C++ so things get difficult for me. 
There is any clue how get Char values over RX/TX.

Appreciate your help.

xoseperez, 2017-10-05:
The SelialLink library just "emancipated" from this repository to its own (https://github.com/xoseperez/seriallink). It's only meant for long int, but of course you can always send a char with is just a byte long (0 to 255). Sending strings would be more difficult with the current implementation but you can always send them in blocks of 4 chars (a long is 4 bytes long).

levipereira, 2017-10-06:
Sorry for this question. 
But any simple example how do that.

Check this example..
Of course this example below will not work.. but how resolv my puzzle on this, with float/char.

##############################
Arduino Uno Code
##############################
# define firmware "2017.10.06.12.47"
const PROGMEM char at_unor[] = "AT+UNOR";   // arduino_uno firwmare release

bool linkGet(char * key) {
if (strcmp_P(key, at_unor) == 0) {
        link.send(key, firmware, false);
        return true;
    }
}
##############################
Esp8266 code
##############################
const PROGMEM char at_unor[] = "AT+UNOR";   // arduino_uno firwmare release

void commsGetFirmware(){
      link.query_P(at_unor);
 }

bool commsSet(char * key, int value) {

    char buffer[50];

   if (strcmp_P(key, at_unor) == 0) {
        int arduino_firmware = value;
       mqttSend (MQTT_Arduino_Firmware, arduino_firmware);
        return true;
   }
}

#################################