xoseperez/sonoffsc #23: cant start flashed esp8266

michaelberg79, 2017-12-29:
I flashed both chips successful without any errors. but if i powering up the device theres happing nothing. 
I inserted a serial print at start of setup(). i cant see any messages in serial monitor.

void setup() {
    // Wait for the ATMEGA to be ready
    Serial.begin(9600);
    Serial.println("firmware started");
    delay(2000);
...

if i am flashing the tasmota firmware the device booting without any problems. what could be the problem?

greets
Michael

michaelberg79, 2017-12-30:
I also flashed espurna successfull. only this firmware is not booting. i dont know how to do.
Do i need to  modify some parameters in the platformio.ini?

michaelberg79, 2017-12-30:
whats the spiffs size i have to choose? the data folder has a size of 75kb. In the platformio.ini is a spifff size of 64 given. maybe this is the reason i cant start the firmware.

michaelberg79, 2017-12-31:
ok i could flash the device with following booting. i inserted the line within platformio.ini 
board_flash_mode = dout
i thought board_flash_mode is the same as flash_mode

Wihanh, 2018-01-11:
I also had this issue, your solution also fixed it for me  thx.