xoseperez/sonoffsc #29: SC v2.0 board_flash_mode should be DOUT

PieBru, 2018-03-03:
Hi, after succesfully flashing the atmega and the esp using "dio" as defined in the original platform.ini, my v2.0 the esp doesn't boot.
However, it boots when flashed using "dout".

I had other minor issues in platform.ini, these settings works here:
```
board = esp01_1m          (in the dev branch it was esp12e)
upload_port = "192.168.4.1"
upload_flags = --port 8266
```

Now at least it boots and logs via UDP, but after some seconds it disconnects my telnet client and it doesn't show the web ui. I will investigate, it seems something screwed up.

Piero

PieBru, 2018-03-03:
As you see, I'm on the dev branch.
Powered via micro USB it reboots every 30 seconds or so, after an exception.
Here is the UDP log when it reboots without loosing too much wifi conn:
```
[SETTINGS] EEPROM size: 4096 bytes
[SETTINGS] Settings size: 65 bytes

[INIT] SONOFFSC 1.2.0
[INIT] xose.perez@gmail.com
[INIT] http://tinkerman.cat

[INIT] CPU chip ID: 0x80A571
[INIT] CPU frequency: 80 MHz
[INIT] SDK version: 1.5.3(aec24ac9)
[INIT] Core version: 2_3_0
[INIT] Core revision: 

[INIT] Flash chip ID: 0x14405E
[INIT] Flash speed: 40000000 Hz
[INIT] Flash mode: DOUT

[INIT] Flash sector size:     4096 bytes
[INIT] Flash size (CHIP):  1048576 bytes
[INIT] Flash size (SDK):   1048576 bytes /  256 sectors
[INIT] Firmware size:       461296 bytes /  113 sectors
[INIT] OTA size:            499712 bytes /  122 sectors
[INIT] SPIFFS size:              0 bytes /    0 sectors
[INIT] EEPROM size:           4096 bytes /    1 sectors
[INIT] Empty space:          16384 bytes /    4 sectors

[INIT] BOARD: SONOFFSC
[INIT] SUPPORT: ALEXA DEBUG_TELNET DEBUG_UDP DOMOTICZ MDNS NTP WEB

[INIT] Last reset reason: Exception
[INIT] Free heap: 32080 bytes
[INIT] Power saving delay value: 10 ms

[TELNET] Listening on port 23
[WEBSERVER] Webserver running on port 80
[MQTT] Async ENABLED, SSL DISABLED, Autoconnect ENABLED
[LIGHTS] Brightness to 255
[LIGHTS] Effect to 12
[LIGHTS] Speed to 128
[LIGHTS] Color to 255
[MAIN] Uptime: 2 seconds
[MAIN] Free heap: 28016 bytes
[MAIN] Time: Not set
[WIFI] Creating access point
[WIFI] MODE AP --------------------------------------
[WIFI] SSID  SONOFFSC_80A571
[WIFI] PASS  fibonacci
[WIFI] IP    192.168.4.1
[WIFI] MAC   6A:C6:3A:80:A5:71
[WIFI] ----------------------------------------------
[MDNS] OK
```

The TX/RX jumpers are connected.
The OTA works.

I don't have an AVR ISP programmer, so no serial and most of the time no 192.168.4.1
Now it seems bricked, after erase_flash the firmware successfully uploads, but it doesn't start flashing with dio or dout mode, even on the master branch.
Any hint?
Piero

xoseperez, 2018-03-04:
I have fixed the platformio.ini file. My SonoffSC was gathering some dust... the log looks good but obviously, there is something going on. Best way to debug it would be to connect it via wire, enableDEBUG_SERIAL_SUPPORT  and monitor the log...

PieBru, 2018-03-04:
Narrowed the problem source, it occurs only in dev branch after enabling the web interface.

Indeed, after disabling almost all the problem appears after enabling WEB_SUPPORT. This is my custom.h
```
#undef USE_PASSWORD
#define USE_PASSWORD            0

#undef DEBUG_SERIAL_SUPPORT
#define DEBUG_SERIAL_SUPPORT    1

#undef DEBUG_UDP_SUPPORT
#define DEBUG_UDP_SUPPORT       1
#undef DEBUG_UDP_IP
#define DEBUG_UDP_IP            IPAddress(192, 168, 4, 2)

#undef HEARTBEAT_INTERVAL
#define HEARTBEAT_INTERVAL      10000      // Interval between heartbeat messages (in ms)

#undef OTA_SUPPORT
#define OTA_SUPPORT             0

#undef TELNET_SUPPORT
#define TELNET_SUPPORT          0

#undef MDNS_SUPPORT
#define MDNS_SUPPORT            0

#undef NTP_SUPPORT
#define NTP_SUPPORT             0

#undef ALEXA_SUPPORT
#define ALEXA_SUPPORT           0

#undef DOMOTICZ_SUPPORT
#define DOMOTICZ_SUPPORT        0

//#undef WEB_SUPPORT
//#define WEB_SUPPORT             0
#undef WEB_MODE_PASSWORD
#define WEB_MODE_PASSWORD       0

#undef LIGHTS_SUPPORT
#define LIGHTS_SUPPORT          0

#define DEBUG_TRACE(module)     {DEBUG_MSG_P(PSTR("[%s@%s] %u\n"), module, __FUNCTION__, __LINE__); delay(100);}
```
As you can see, I added LIGHTS_SUPPORT and OTA_SUPPORT to cut out the corresponding functions.
Enabling/disabling WEB_PASSWORD doesn't change the problem.

I will soon investigate further, as this problem occurs in dev but not in master.
Meanwhile, do you think in dev branch there are additions that can cause this behaviour?

PieBru, 2018-03-06:
No exception 9 by disabling WEB_EMBEDDED in dev branch, but also no home page because I didn't search how to update html in spiffs.
However, WEB_EMBEDDED was enabled also in the main branch, so the cause of this blocking issue may lie with the additions to web.ino in the dev branch (_on* callbacks, _authenticate(), etc).
It's not easy for me to proceed in this problem solving in a reasonable time, sorry.
Piero

xoseperez, 2018-03-06:
Thank you very much @PieBru . I'll try to spend some time on this over the week.

derFrickler, 2018-11-28:
Managed to flash the dev version to my SonoffSC 2.0, but now when i try to login to the website i either get a wrong credentials or the device crashes with Exception 9.

Any hints here?