xoseperez/sonoffsc #30: Problem upload firmware ATMEGA

Symbyos83, 2018-04-02:
Hello, in the meantime I wanted to congratulate you for the excellent development. I wanted to ask you about the firmware upload on the ATMEGA. on the ESP no problem, while when I go to load on ATMEGA with platformio and USBASP (modifying the parameters on the platformio file, I find myself this error "avrdude: verification error, first mismatch at byte 0x0001
avrdude: verification error; content mismatch
 0x10 != 0x94
avrdude: safemode: Fuses OK (E:FD, H:DE, L:FF)", 

what can it be? I banged my head everywhere now.
Thank you
Andrea

xoseperez, 2018-04-02:
I recall getting this error in the verification phase when `avrdude` compares contents *beyond* the image size. Solution in these cases is to force erasing the memory with the `-e` option.
You can also try to use the bootloader in the ATMega and flash it with a USB2UART if you have one.

xoseperez, 2018-04-02:
You can add this to the platformio env:

```
upload_flags = -Pusb -e
```

Symbyos83, 2018-04-03:
so I have to insert that line on the platformio.ini file of the ATMEGA after the following lines
  [Platformio]
env_default = sonoffsc
src_dir = sonoffsc

Sorry but I'm not very practical with Atom and Platformio.

Symbyos83, 2018-04-03:
@xoseperez I managed to flash the ATMEGA. I reconnected the bridges between AMEGA and ESP, connect the USB but I can not find any network Do I have to flash the esp with the -e option?

xoseperez, 2018-04-03:
Change the platformio.ini file in the atmega folder to this:

```
[platformio]
env_default = sonoffsc
src_dir = sonoffsc

[env:sonoffsc]
platform = atmelavr
board = uno
framework = arduino
upload_protocol = usbasp
upload_flags = -Pusb -e
build_flags = -DDHT_TYPE=DHT22
lib_deps =
     Adafruit Unified Sensor
     DHT sensor library@1.3.0
     Adafruit NeoPixel@1.1.2
     https://github.com/kitesurfer1404/WS2812FX#069139b
```

Symbyos83, 2018-04-03:
yes, the ATMEGA it's ok now.. But i have a problem with ESP. Not show wifi to connect, no blink led... nothing. i have sonoff sc v2.0!!!

Symbyos83, 2018-04-03:
Resolved!!!! put board_flash_mode = dout

closed 2018-04-04 by Symbyos83