xoseperez/sonoffsc #38: Replacing the DHT with a BME680

rbm78bln, 2018-11-02:
In you blog-posting about the Sonoff SC you were mentioning to better replace the DHT11 by a DHT22, @xoseperez 

Would it be possible to "replace" the DHT11 also by a [BME680](https://www.aliexpress.com/item/BME680-temperature-and-humidity-temperature-pressure-high-altitude-sensor/32909028283.html) by using the ISP connector of the atmega328?

This would massively empower the Sonoff SC for just ten bugs and with almost no effort.

xoseperez, 2018-11-02:
Hi

It is possible but you will have to do some changes in the wiring and the
firmware, of course. The DHT11/22 use one digital IO but the BME680 is a
I2C device and it needs two digital IO. The Atmega328 has a single hardware
I2C and I think it's not available, so you will have to use a softwareI2C
library to use two random IO to connect the BME. Then you will probably
have to modify the BME680 library to use that I2C library instead of the
default Wire lkbrary.

So, yes it's possible but it's somewhat involved...

rbm78bln, 2018-11-02:
Oh yes, that's right!
I was especially thinking about the "air quality rating" feature of that sensor :-D

closed 2020-08-17 by rbm78bln